SFUSatClub/CubeSat-Radio #3: Need ability to probe desired points in RF path

Alenux55, 2017-03-31:
Add a vertical SMA population option between CC1125 and balun, balun and switch, and switch and front end

Alenux55, 2017-03-31:
Also, make this a resistor population option, like a T-junction

TobiNakamura, 2017-05-25:
Can the T-junction be made such that the placement of SMA can be selected facing towards or away from the font of the signal chain? Should there be 2 SMA footprints at each node in the signal chain; one for each direction?

Alenux55, 2017-05-31:
Instead of one SMA for each direction, use a single 0402 SMA pad such that a jumper can be placed at 45 degrees in either direction:
[image]

TobiNakamura, 2017-06-10:
[image]
Too weird?

Alenux55, 2017-06-11:
What if we just make the rectangular pad into a square and rotate it 45 degrees?

Alenux55, 2017-06-11:
Lets move this to the issue in KiCad-Lib since the main objective of this issue is complete. [https://github.com/SFUSatClub/KiCad-Lib/issues/13](https://github.com/SFUSatClub/KiCad-Lib/issues/13)

closed 2017-06-11 by Alenux55