SFUSatClub/CubeSat-Radio #9: Incorporate connector ESD connection

Alenux55, 2017-03-31:


TobiNakamura, 2017-05-18:
Also for tactile switches and toggle switches and any other exposed interfaces

TobiNakamura, 2017-06-10:
No ESD protection on CC debug connector or PC104 connector due to large number of pins. ESD suppression was placed on test points, jumpers, and potentiometers. 2d7826e3472c9caad9861fabc356ae1afd1395ca

Alenux55, 2017-06-11:
- Are we relying on the LDOs as ESD protection from Vin?
- Do we need D10 if we have D9?

TobiNakamura, 2017-06-11:
Vin is 5V, the knee of the TVS diodes are 5V or so. Yes, too cheap to get another TVS diodes, the LDO can burn. TVS should be on SKY input and on POT to protect 3v3 line.

Alenux55, 2017-06-11:
Gotcha, because **R7** is close to **RV1**:
[image]

And then none on **R8** because it's far from the jumper. Do we want any extra protection on P4 as it's attached to the debugger?

TobiNakamura, 2017-06-11:
TVS diode added OP_3v3. 39e230a1014d43e301bcf52a7b34e85acb464f90

closed 2017-06-12 by TobiNakamura