SFUSatClub/CubeSat-Radio #13: 2N7002DW pin assignment incorrect

TobiNakamura, 2017-04-20:
Component pin assignment is incorrect. Please read datasheet https://www.diodes.com/assets/Datasheets/ds30120.pdf

TX and 3V6 indicator LEDs do not work because of this.

TobiNakamura, 2017-05-26:
Fixed, please review

liquiddandruff, 2017-05-26:
Can also paste commit hash to reference the commit too ;)  e4fd6992

Alenux55, 2017-05-31:
[image]
Part is mirrored. Please update symbol and make footprint mapping as follows:
6  5  4
1  2  3

TobiNakamura, 2017-05-31:
ef6c72adf7fbf4c4be107657a5a16833a87c14b3

Alenux55, 2017-05-31:
Checked, looks good
[image]

closed 2017-05-31 by Alenux55