SFUSatClub/CubeSat-Radio #20: Move POT for VPC control to ground, fixed resistor to 3v3

TobiNakamura, 2017-05-31:
Fixed resistor should be 3.2K, POT will be the same one used for Rev_1.
This way turning the POT full way will result in 2.5V or 0V.
Considered 50K pull down present in SKY65366 for caluclation.

TobiNakamura, 2017-06-07:
764d84e27edea83bda7cb37d0cbc6eef1bb36361

Alenux55, 2017-06-07:
Potentiometer is 10 kΩ, so from the [datasheet](http://www.bourns.com/docs/Product-Datasheets/TC33.pdf), the minimum resistance is **200 Ω**:
[image]

So we have:
_Vout,max = [(3.3 V)*(10000)]/[(3200+10000)] = **2.5 V**_
_Vout,min = [(3.3 V)*(200)]/[(3200+200)] = **0.2 V**_

Looks good. I doubt we'll need true **0 V** considering this is for the PA (will most likely want that value high)

closed 2017-06-07 by Alenux55