SFUSatClub/CubeSat-Radio #23: Check Connector Labeling

TobiNakamura, 2017-06-09:
Check labels on jumpers, SMA connectors, PC/104 connector, test points.
I forgot to label the LEDs.

TobiNakamura, 2017-06-09:
938609b44109fadba14369935545f2254597ae9c

Alenux55, 2017-06-09:
- **W3** looks good on the **3.6 V** line
- **P3** looks good
    - Can we shorten the silkscreen from _AMP EN_ to _EN_? I think the text above the part conveys that it controls the amplifier
- **W2** looks good, but wouldn't it be better to connect to the center pin of P2 so we can check VPC from either source?
- **P2** looks good

[image]

- **P4** looks good
- **W4** looks good
- **W1** looks good
    - Need to make sure we can configure GPIO2 of CC1101 to do this (see **table 41** setting **0x1C** in the datasheet for candidate; active low signal, active for **SLEEP** and **RX** states)

[image]

- **J8** looks good
- **J9** looks good

[image]

TobiNakamura, 2017-06-10:
7a60c29a94ea7f9e260f3175538d6a2e0feaa4c7

Alenux55, 2017-06-11:
Requested changes look good, closing
[image]

closed 2017-06-11 by Alenux55