SFUSatClub/CubeSat-Radio #25: Check RF path

TobiNakamura, 2017-06-10:
[image]
Check CPWG, passives, wall of ground vias.

TobiNakamura, 2017-06-10:
The substrait thickness is not 0.2mm, its 1.6mm.
The track width is a bit worrisome.
[image]

TobiNakamura, 2017-06-11:
[image]

TobiNakamura, 2017-06-13:
PCBWay confirmed that Dk is 3.8.
[image]

Alenux55, 2017-06-14:
Do we want to update the trace width to get it closer to **50 Ω**? At the current thickness, we'll get **52.5 Ω**

Alenux55, 2017-06-14:
Also, do we want to add fencing vias along the RF traces? According to [this document](www.edn.com/Pdf/ViewPdf?contentItemId=4406491), if the vias are spaced less than about **2 inches**, we won't run into ground impedance issues.

Further, from the same document:
_Spacing the ground vias at 1/8th of a wavelength gives good isolation here also, but for the ultimate
isolation available (i.e., greater than the 120 dB that a modern network analyzer can measure) it has
been found that the via spacing should be kept to 1/20th of a wavelength or less._

In our case, **1/20** of the wavelength is about **3.5 cm**, which is easily doable.

Alenux55, 2017-06-14:
Never mind about the via fencing

TobiNakamura, 2017-06-14:
RF traces are now 0.6mm width with 0.2mm spacing to ground.

closed 2017-06-14 by TobiNakamura