SFUSatClub/CubeSat-Radio #39: RF Switching

TobiNakamura, 2018-03-03:
Currently a the architecture is
antenna --- diplexer --- Switch --- PA
                        |                  |
                Upper band     LNA

This introduced ~0.5db of attenuation at the diplexer and a further 0.5db at the switch. If we eliminate the switch and use a SP4T RF switch we can reduce the total loss to 0.5db

TobiNakamura, 2018-03-05:
Even if we dont go with this architecture we still need a 2W rated SPDT switch

TobiNakamura, 2018-03-20:
@Alenux55 added SKY13345-368LF in conjunction with CD4555B to replace the diplexer.
[image]

TobiNakamura, 2018-03-20:
@Alenux55 What do you think about using Q3 as an error output?

Alenux55, 2018-03-20:
Like to display it on an LED?

Alenux55, 2018-04-17:
@TobiNakamura I am going to close this unless we can think of something to use for the extra outputs

closed 2018-04-20 by Alenux55