SFUSatClub/CubeSat-Radio #53: Remove OP_3v3 jumper

TobiNakamura, 2018-04-23:
[image]
This feature make the power plane more complex. We can remove it and connect VDD to the main 3v3 rail. If the debugger is to provide power you can apply it directly to the 3v3 pin in the PC104 bus.

Alenux55, 2018-04-24:
Are we able to turn off the debugger's output power? If not, do we anticipate never needing to control the board with the CC debugger and powering it from the PC104 stack?

TobiNakamura, 2018-04-24:
http://www.ti.com/lit/ug/swru197h/swru197h.pdf
Schematic on appendix A. Pin 9 of the CC-Debugger header is permanently connected to 3v3.
I think the only use case for the CC-debugger is initial bring-up. So probing the CC1101 output.

TobiNakamura, 2018-04-26:
Pin 9 (OP_3.3) has been disconnected. 3.3v net of CC1101.sch now connected to board wide 3V3 net.

closed 2018-04-26 by TobiNakamura