SFUSatClub/CubeSat-Radio #54: Manual Gain Control

TobiNakamura, 2018-04-26:
Gain control using the potentiometer should vary the output between 0 and 3.1V.
This can be done linearly using only a potentiometer, however we are only interested in using the circuit with an output above ~2V. A non-linear circuit can be constructed to maximise the resolutions around the desired output voltage.
[image]
Note R411 and R412 exists to replace the potentiometer in case a fixed value is desired.
[image]
The x-axis is the fraction of wiper position along the length of the rail, y is the output voltage. The non-linearity of the output can be controlled using R_2. I suggest we use 100k ohms.

TobiNakamura, 2018-04-27:
100k 0402 res added

Alenux55, 2018-09-20:
Done.

closed 2018-09-20 by Alenux55