SFUSatClub/CubeSat-Radio #55: Upper Band Balun and Filter Needs Redesign

TobiNakamura, 2018-05-04:
Currently we are using Johanson Tech. 0915BM15A0001E integrated balun and filter (Center freq 915MHz). The required center frequency is 790MHz.

Alenux55, 2018-09-27:
IIRC we want the center frequency around 790 MHz

Alenux55, 2018-10-02:
How about [0783FB15A0100E](https://www.digikey.ca/products/en?keywords=712-1537-1-ND) filterbalun and match it to the CC1101? See CC1101 [Design Note DN017](http://www.ti.com/lit/an/swra168a/swra168a.pdf) as well:
[image]

TobiNakamura, 2018-10-02:
Good idea.
Per [DN017 (swra168a)](http://www.ti.com/lit/an/swra168a/swra168a.pdf) the differential impedance at the CC1101 port looking towards the antenna is  Z= 86.5 + j43 Ohm at 868/915 MHz.

Alenux55, 2018-10-02:
I will see if TI happens to have the impedance data for 700-800 MHz

Alenux55, 2018-10-04:
[TI post](https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/156/t/733964)

Alenux55, 2018-10-08:
TI suggests two alternative filter baluns, the claim is that we only lose 0.25 dB extra gain to insertion loss